illiliti/libudev-zero #4: lvm2

muevoid, 2020-09-13:
lvm2 does not build with the following errors.

```
/usr/bin/ld: ../../libdm/libdevmapper.so: undefined reference to `udev_queue_get_udev_is_active'
/usr/bin/ld: ../../libdm/libdevmapper.so: undefined reference to `udev_queue_unref'
/usr/bin/ld: ../../libdm/libdevmapper.so: undefined reference to `udev_queue_new'
```

It may be user error any help would be appreciated.

illiliti, 2020-09-13:
According to https://www.freedesktop.org/software/systemd/man/libudev.html - udev_queue is deprecated

_Furthermore, libudev also exports legacy APIs that should not be used by new software (and as such are not documented as part of this manual). This includes the hardware database known as udev_hwdb (please use the new sd-hwdb(3) API instead) and the udev_queue object to query the udev daemon (which should not be used by new software at all)._

I did an exception only to udev_hwdb because many software has mandatory dependency on it. AFAIK only lvm2 has dependency on udev_queue, which means i'm not going to do an exception. As a workaround i suggest you to compile lvm2 without udev support(honestly, udev support in lvm2 is senseless)

closed 2020-09-13 by illiliti

muevoid, 2020-09-13:
Yes I forgot I opened this issue the community build for lvm2 used pkg config to detect udev. I disabled it thank you!