illiliti/libudev-zero #8: pkgconfig file is rather messed up

neheb, 2020-11-03:
Normal pkgconfig files have the following format:

prefix=/usr
exec_prefix=/usr
libdir=${exec_prefix}/lib
includedir=${prefix}/include

This one is lacking the prefix variable which interferes with OpenWrt (we need to override the prefix).

E5ten, 2020-11-03:
I can probably make a PR to fix this later when I'm at my computer, but just to note how I think it would be fixed, prefix and exec_prefix variables would be added to the .pc, and substitutions for each would be added to the sed used to generate the file. Then a check would be done to see if the LIBDIR/INCLUDEDIR start with PREFIX, and if they do, strip the leading prefix and prepend `\${prefix}` or exec_prefix accordingly, then use whatever shell variables were set to this in place of the make variables in the build rule.

neheb, 2020-11-03:
My basic solution https://github.com/openwrt/packages/blob/b443dda1f330a6b947d5dd287c4115a63713ba67/libs/libudev-zero/patches/010-pkgconfig.patch

Works well enough.

E5ten, 2020-11-03:
It won't work if someone sets LIBDIR or INCLUDEDIR to a path not starting with PREFIX.

PR #9 (2020-11-11): libudev.pc: add prefix and exec_prefix support
Should fix #8
closed 2020-11-11 by illiliti