illiliti/libudev-zero #17: libudev-zero crushes sway upon using x11 application

keleset, 2021-02-13:
So, everything that uses wayland works perfectly fine.
As soon as I open any app using x11 (xwayland) - sway crashes.
I tried to debug the issue before, but sway verbose logs and even gdb connected to sway provide nothing meaningful.
The only one fact I have - the issue is not present if I switch to eudev.
Setup - Alpine on laptop with integrated and dedicated AMD graphics.

illiliti, 2021-02-14:
Does the issue persist if you downgrade to previous version 0.4.7?

keleset, 2021-02-14:
Yes, it does. And even before that, I've previously tried libudev-zero, but didn't have time to debug it and quickly gave up back then, not sure what version it was...

illiliti, 2021-02-16:
Can you apply this PR https://github.com/swaywm/wlroots/pull/2626 to wlroots and then trigger the crash?

keleset, 2021-02-16:
I've built master branch of sway & wlroots and issue seems to be gone now. Guess that was a sway/wlroots bug after all.

closed 2021-02-16 by keleset