illiliti/libudev-zero #18: dhcpcd: no valid interfaces found

git-bruh, 2021-02-15:
When `dhcpcd` is started by the runit (`busybox` init) service, it says `no valid interfaces found` and just stops there, the service has to be restarted manually for it to detect my wifi adapter, but this is not the case with `eudev`:

```
no such user dhcpcd
dhcpcd-9.4.0 starting
dev: loaded udev
DUID 00:01:00:01:27:5e:0a:a4:d0:39:42:49:f3:14
no valid interfaces found
no interfaces have a carrier
```

I'm not sure if this is directly related to `libudev-zero`, but might be related to `dhcpcd` starting before `mdev` is initialised, any ideas ?

illiliti, 2021-02-16:
> might be related to `dhcpcd` starting before `mdev` is initialised

This. Technically, libudev-zero should signal dhcpcd about new devices via hotplugging. Ensure that you have properly working hotplugging support.

There is also a possibility of race condition between dhcpcd and mdev which can lead to impossibility to detect net devices due to permission issues. Try to run dhcpcd as root or implement proper synchronization to mdev.

git-bruh, 2021-02-19:
Hmm, hotplugging seems to be working fine, `dhcpcd` is running as root.

> implement proper synchronization to mdev

How does one go about this ?

illiliti, 2021-02-19:
> How does one go about this ?

Ah, sorry I forgot about that dhcpcd should be always run as root. You don't need it. This synchronization is needed for e.g. root-less Xorg which is usually(or rarely, it depends on init implementation) fails to initialize drm devices at startup due to access rights. I'll probably add some docs about it to readme.

Does the dhcpcd working if compiled without udev support?

git-bruh, 2021-02-20:
Just compiled it without udev support, works great now.

git-bruh, 2021-02-23:
@illiliti Do you plan to tackle this in `libudev-zero` ? If not, I'll close the issue.

illiliti, 2021-02-23:
Yes, I'll investigate this further later

closed 2021-05-11 by illiliti

git-bruh, 2021-05-11:
Yeah it stopped happening for me, maybe it was an issue with my old config