illiliti/libudev-zero #19: bluez support

keleset, 2021-02-16:
Is it possible to make libudev-zero work with bluez or maybe develop some instructions to work it around? Bluetoothctl is unable to find any bluetooth controllers.

illiliti, 2021-02-16:
Bluez uses `udev_hwdb` API which is not yet implemented. Possible workaround is compiling bluez without udev support.

```
./configure --disable-udev
make
```

keleset, 2021-02-16:
bluez still seem unable to detect bluetooth controller, even if built with `--disable-udev`

illiliti, 2021-02-19:
Can you send me exact error with and without libudev-zero?

keleset, 2021-02-26:
There is no error actually, I'm just unable to detect any bluetooth adapters with bluez, even though `rfkill list` sees them perfectly fine.

keleset, 2021-02-26:
A bit more research revealed that a firmware file for the bt device was missing, fixed now.
BTW, thousand kudos for libudev-zero! Thing is, actually, an outstanding replacement for udev, great job!

closed 2021-02-26 by keleset