illiliti/libudev-zero #23: Hotplugged devices are not being detected by X

TAAPArthur, 2021-05-10:
So all my devices are detected and function normally when they are known upon starting X. However, if I attempt to hot plug a device, it is not recognized. Similarly, when I unplug a device, X doesn't notice its disappearance. Restarting X allows the new devices to be correctly detected, so seems like an issue with hotplugging.

I'm using the uevent method. I'm using a custom device manager, but it is similar to smdev. And in regards to uevent files, the same as the helper scripts provided in that it just dumps the environment to /tmp/.libudev-zero/uevent.XX. The uvent files are generated correctly. I also using the provided uevent helper scripts to no avail.

I'm using the latest libudev-zero (6685cfa). Made sure I was at the latest version of recompiled libinput and xf86-input-libinput. Also undid any patches I made to X. 

Interestingly enough, I have Xorg in debug mode and I see messages like `input-thread: InputThreadDoWork waiting for devices` when I unplugging/re-plugging devices

illiliti, 2021-05-12:
Hotplugging just works for me without any issues. I'm using xorg-server 1.20.11 and busybox mdev with provided contrib/mdev.conf + some modifications. Are you sure that kernel is properly configured? Permissions of /dev/* are set correctly? Are you using modular(external modules) or monolithic(builtin modules) kernel?

TAAPArthur, 2021-05-12:
I can confirm that the uevent helper is being triggered so assume the kernel is fine.
Running x as a non-root users who is part of a group that has read/write access to relevant /dev/ files. Also if it was a permission issue, I'd always have an issue and not just with hotplugging. Did try running X as root and that didn't help.
Some combination. The ones from these input devices are modules which are loaded currently shortly after boot. Can confirm the modules get loaded correct as the devices are at one point detected and usable.

I'm sure it is something I have configured somewhere... But I seem to have issues like this often where a given device isn't behaving as expected (other cases where my devices being detected as the wrong type and ignored by X). May be useful to have some debug logging? Like `device {name} added/removed` or `device {name} detected as DEVICE_TYPE`

illiliti, 2021-05-13:
Please attach the xorg log and uevent files.

> But I seem to have issues like this often where a given device isn't behaving as expected

What do you mean?

> May be useful to have some debug logging? Like `device {name} added/removed` or `device {name} detected as DEVICE_TYPE`

I'll think about it.

phkrl, 2021-05-15:
I also encountered this while testing in qemu. I started X in VM and then added usb-mouse via device_add. The proper node appeared in /dev, uevent was created and was detected by inotifywait but X didn't detect it until restart

TAAPArthur, 2021-05-15:
@illiliti Attached [Xorg log](https://github.com/illiliti/libudev-zero/files/6486756/Xorg.0.log.txt). After starting X, waited a while unplugged/replugged devices (no change in the logs) and then restarted X. I turned off debug logging because that seemed to be too verbose and contain irrelevant info other than the `input-thread: InputThreadDoWork waiting for devices` which show up when the device is removed.

The uvent files are just a dump of the environment. The `SUBSYSTEM` is `input` and the `DEVNAME` is `input/event*` and the `ACTION` is either `add` or `remove`. `SEQNUM`, `MINOR` and `DEVPATH` tend to also be set

> > But I seem to have issues like this often where a given device isn't behaving as expected
>
> What do you mean?

For the other commits I've submitted, had a similar problem of not knowing where the disconnect was between libudev-zero and X. In those cases it was because the devices weren't tagged correctly, but the X log just said X was ignoring the device. So it wasn't immediately clear what the problem was.

TAAPArthur, 2021-05-16:
@illiliti 
Think I found the problem area

udev_monitor.c
```
    while (epoll_wait(udev_monitor->efd, epoll, 2, -1) != -1) {
        for (i = 0; i < 2; i++) {
            if (epoll[i].data.fd != udev_monitor->pfd[0]) {
                continue;
            }

            read(udev_monitor->pfd[0], data, 1);
            write(udev_monitor->pfd[1], "1", 1);
            pthread_barrier_wait(&udev_monitor->barrier);
            return NULL;
        }

        len = read(udev_monitor->ifd, data, sizeof(data));

        if (len == -1) {
            continue;
        }

        for (i = 0; i < len; i += sizeof(struct inotify_event) + event->len) {
            event = (struct inotify_event *)&data[i];

            // TODO directory is removed
            if (event->mask & IN_IGNORED) {
                break;
            }

            send(udev_monitor->sfd[1], event->name, event->len, 0);
        }
    }
```
Seem to be stuck in `epoll_wait` and don't reach `read(udev_monitor->ifd, data, sizeof(data));` Replacing the while condition with while(1) and skipping the inner for loop, allows for hotplugging.

Upon further investigation, it seems like epoll_wait is returning -1 (I imagine errno=EINTR). I'll send fix shortly

phkrl, 2021-05-16:
> @illiliti
> Think I found the problem area
>
> udev_monitor.c
>
> ```
>     while (epoll_wait(udev_monitor->efd, epoll, 2, -1) != -1) {
>         for (i = 0; i < 2; i++) {
>             if (epoll[i].data.fd != [...]

Just tested your patch with my qemu setup, it didn't work.
Uevents are created, so helper works as expected.

TAAPArthur, 2021-05-16:
@phkrl "it didn't work" is very descriptive. My patches solves the issue I was having so maybe we just had similar symptoms? 

Are the inotify events being generated for you (or do you even reach the line where they are read)? Can you confirm that the you were exiting this while loop prematurely and now aren't?

phkrl, 2021-05-17:
I found the problem: in my case it was caused by permissions of uevents created by mkstemp (i used helper.c). If i set permissions to 644, hotplug works

illiliti, 2021-05-18:
@phkrl i added extra `fchmod(2)` call to ensure that file is readable.

TAAPArthur, 2021-05-27:
@illiliti Just tested HEAD and I still have the same hotplugging issue.

illiliti, 2021-05-27:
This is very odd. I'm still trying to understand what is the root of problem? I can't figure that out because i can't reproduce it.

@TAAPArthur please try to debug it or provide more useful information.

TAAPArthur, 2021-05-27:
@illiliti I get that my last message wasn't the most useful. But from my standpoint, I had provided a complete fix that was quietly ignored in favor of the current solution, which doesn't solve the problem. Now I'm being asked to further debug why the new "solution" didn't work? 
Was there something wrong with the original fix that I have tested and confirmed works?

commit f2083f0e (2021-05-25): udev_monitor.c: block signals within event loop

Fixes #23
Closes #24
closed 2021-06-01 by illiliti