illiliti/libudev-zero #33: Standalone volume/power buttons are no longer treated as keyboards

TAAPArthur, 2021-06-18:
I have a PinePhone that has volume buttons as one device an a power button as another. They should be treated as keyboards so they can be associated with XF86AudioLowerVolume, XF86AudioRaiseVolume and XF86XK_PowerOff. 

Proposing to revert e23633ec562dc781a543c8e2b6ed0007a86e7158. The commit message didn't go into detail about what kinda of device it was trying to fix so don't know if such devices exists or this was just a theoretical fix. I can confirm that the range we iterate over from [KEY_ESC, BTN_MISC) contains `KEY_VOLUMEDOWN`, `KEY_VOLUMEUP`, `KEY_POWER`. Haven't looked into exactly what key bits are directly detected for my device, but can confirm that reverting e23633ec562dc781a543c8e2b6ed0007a86e7158 fixes the problem.

illiliti, 2021-06-18:
If we revert that commit, every device that has only BTN_* will be treated as keyboard(which is wrong). The problem is that kernel/evdev doesn't distinguish buttons and keys. If device has EV_KEY, we can't assume that this is a keyboard and/or just device with keys. The only way to confirm that device has keys is to check bits within KEY_*.

Can you post output of this programs?

1. https://termbin.com/gh5g8
2. https://github.com/freedesktop/evemu
```
evemu-describe /dev path to device with volume buttons
evemu-describe /dev path to device with power button
```

TAAPArthur, 2021-06-18:
>If we revert that commit, every device that has only BTN_* will be treated as keyboard(which is wrong).

Fair point. I see my normal 3 button mice being detected as keyboards.

I can provide data from evtest
```
Input driver version is 1.0.1
Input device ID: bus 0x19 vendor 0x1 product 0x1 version 0x100
Input device name: "1c21800.lradc"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 114 (KEY_VOLUMEDOWN)
    Event code 115 (KEY_VOLUMEUP)
```
```
Input driver version is 1.0.1
Input device ID: bus 0x0 vendor 0x0 product 0x0 version 0x0
Input device name: "axp20x-pek"
Supported events:
  Event type 0 (EV_SYN)
  Event type 1 (EV_KEY)
    Event code 116 (KEY_POWER)
Properties:
```

For the pastbin (filtered out all irrelevant devices):
```
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - ID_INPUT - 1
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - DEVNAME - /dev/input/event0
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - MINOR - 64
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - MAJOR - 13
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - SYSNUM - 0
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - SYSNAME - event0
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - SUBSYSTEM - input
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - DEVPATH - /devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0 - SYSPATH - /sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/event0
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - ID_INPUT - 1
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - DEVNAME - /dev/input/event1
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - MINOR - 65
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - MAJOR - 13
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - SYSNUM - 1
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - SYSNAME - event1
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - SUBSYSTEM - input
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - DEVPATH - /devices/platform/soc/1c21800.lradc/input/input1/event1
/sys/devices/platform/soc/1c21800.lradc/input/input1/event1 - SYSPATH - /sys/devices/platform/soc/1c21800.lradc/input/input1/event1
```
I also tried logging `test_bit(key_bits, KEY_POWER)`, `test_bit(key_bits, KEY_VOLUMEUP)`, `test_bit(key_bits, KEY_VOLUMEDOWN)` and they all came back negative.

illiliti, 2021-06-19:
Thanks.

Post here content of these files:
```
/sys/devices/platform/soc/1c21800.lradc/input/input1/uevent
/sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/uevent
```

TAAPArthur, 2021-06-20:
cat /sys/devices/platform/soc/1c21800.lradc/input/input1/uevent
```
PRODUCT=19/1/1/100
NAME="1c21800.lradc"
PHYS="sun4i_lradc/input0"
PROP=0
EV=3
KEY=c000000000000 0
MODALIAS=input:b0019v0001p0001e0100-e0,1,k72,73,ramlsfw
```
cat /sys/devices/platform/soc/1f03400.rsb/sunxi-rsb-3a3/axp221-pek/input/input0/uevent
```
PRODUCT=0/0/0/0
NAME="axp20x-pek"
PHYS="m1kbd/input2"
PROP=0
EV=3
KEY=10000000000000 0
MODALIAS=input:b0000v0000p0000e0000-e0,1,k74,ramlsfw
```

commit bf23c9e5 (2021-06-25): udev_device.c: fix integer overflow

Bitwise AND operation can produce value that may be larger than
test_bit() return value. Convert result to bool using `!!` to
avoid integer overflow.

Fixes: #33
closed 2021-06-25 by illiliti