illiliti/libudev-zero #34: don't install static library (by default)?

Hello71, 2021-07-01:
over at alpine linux, we recently had some issues relating to improperly mixing libraries linked against libudev-zero static library with eudev dynamic library. this caused us quite a bit of grief, because in certain circumstances this issue is not diagnosed by static or dynamic linker, and explodes very confusingly at runtime due to totally incompatible data structures.

therefore, I would like to suggest removing the static library or at least not installing it by default. I think it is a major footgun for users, and the overhead of loading and calling a separate dynamic library is negligible, especially compared to actually probing and interacting with devices.

thanks!

illiliti, 2021-07-02:
This is unfortunate situation with alpine, but I don't think I can do anything on my side. It's common practice to produce static and shared library by default(eudev and other libraries does that too). I also don't understand where did you see footgun here.

Hello71, 2021-07-02:
hm, my mistake. i think it should still not be installed, but it should be made at distro level. thank you for your input.

closed 2021-07-02 by Hello71

E5ten, 2021-07-03:
Maybe the Makefile could be changed to have a mechanism for determining if it should install the shared, static, or both libraries? E.g. 2 different install targets, or SHARED and STATIC variables that are enabled by default but when set to something (maybe an empty string?) don't install their corresponding library?

illiliti, 2021-07-04:
This is unneeded complexity because I don't like the idea of doing something conditionally in terms of Makefile. I think Makefile should stay immutable. Current Makefile is very readable and you can easily patch it to remove shared/static[0] library.

https://github.com/glaucuslinux/cerata/blob/2636c22727d1256a020fd85bc8c942d475b534a0/libudev-zero/patches/glaucus/0001-disable-static.patch