illiliti/libudev-zero #38: Adding version symbol to Makefile

Rio6, 2021-09-06:
Right now when using some libraries like libusb, the dynamic linker would complain about missing version symbols (but the program still works fine). For example when using adb:
```
$ adb
adb: /usr/lib/libudev.so.1: no version information available (required by /usr/lib/libusb-1.0.so.0)
Android Debug Bridge version 1.0.41
Version 31.0.2-android-tools
Installed as /usr/bin/adb
```

I'm not sure how libudev-zero map to systemd versions, but if we can add a file like this https://github.com/systemd/systemd/blob/main/src/libudev/libudev.sym and using it when linking we can get rid of the warning.

Rio6, 2021-09-06:
An empty `LIBUDEV_183 {};` seems to work too for my case with adb.

closed 2021-09-08 by illiliti

TAAPArthur, 2021-11-16:
@illiliti Just a heads up that 000ff7b broke complication with tcc
```tcc: error: unsupported linker option '-version-script,libudev.sym'```
Which is simply enough for the user to resolve. But think breaking complication outweighs the benefits of silencing a warning.

Rio6, 2021-11-16:
as mentioned in https://github.com/illiliti/libudev-zero/pull/40#issuecomment-939492896, arcan requires symbol versioning to be there, so it's not just a warning in some cases.

illiliti, 2021-11-16:
Reverted. Trying to catch up with upstream is not good idea here

commit 000ff7bf (2021-09-08): Makefile: add stub symbol versioning

Fixes: #38