illiliti/libudev-zero #50: The capital importance of this project

vanfanel, 2022-01-06:
Hello @illiliti 

I found this project only after a few days experimenting with UDEV alternatives, which took me to BusyBox's MDEV, which in turn lead me to a libudev alternative to use along with MDEV.. and here I am.

I have just cloned the repo, and will build this piece of software today.

Just wanted to express how important this is: UDEV is now compromised because of the user-hostile people and model behind it's development, EUDEV is dead... and libudev needs one or the other, so this project needs to live on so MDEV can be a libudev alternative in the future.

Thanks for trying, thanks for fighting these SystemD/UDEV anti-user corporations.
This is not an issue and can be closed. Just wanted you to know that I know.

l29ah, 2022-03-19:
https://github.com/eudev-project/eudev eudev is alive

vanfanel, 2022-03-19:
@l29ah So, does that mean that eudev is going to be developer further into the future now???? :O

gordon-quad, 2023-09-27:
@l29ah pretty much dead https://bugs.gentoo.org/913702