illiliti/libudev-zero #51: First usage question: Removing UDEV/EUDEV?

vanfanel, 2022-01-06:
Since this is a drop-in replacement for libudev/libeudev, should I purge libudev/libeudev before installing this on the system?

The problem is that Devuan seems to heavily depend on libeudev, and purging it will purge fudamental parts of the system too:

```
The following packages will be REMOVED:
  apt* apt-transport-https* apt-utils* d-shlibs* dmsetup* eudev* fuse* i2c-tools* init* initramfs-tools* initramfs-tools-core*
  libapt-pkg6.0* libdevmapper1.02.1* libeudev-dev* libeudev1* libfido2-1* libparted2* libpci3* libusb-1.0-0* mount* ntfs-3g*
  openssh-client* openssh-server* openssh-sftp-server* parted* pciutils* rpi-eeprom* rpi-eeprom-images* sysvinit-core* usbutils*
  util-linux*

```

So, any notes on how to proceed?

phkrl, 2022-01-14:
I guess you can build libudev-zero as deb package providing libeudev. This is how i made it work in Artix

vanfanel, 2022-01-14:
@phkrl: That's a GREAT idea! Hadn't though about that possibility... I will take a look.

phkrl, 2022-01-14:
> Do you have instructions for that? ie: building a package that provides another package...
It entirely depends on package building mechanism. I never built deb packages, for Artix (which uses Arch format) supplying provides=(...) in PKGBUILD is enough

vincele, 2024-06-14:
@vanfanel you may even want to add a `conflicts` to your `provides`, see:
https://www.debian.org/doc/manuals/debian-faq/pkg-basics.en.html#depends