illiliti/libudev-zero #53: [BUG] libudev-zero overwrites eudev, libnm linkage break

Kochise, 2022-03-22:
Hi, working on an Alpine package :

https://github.com/ElektraInitiative/libelektra/issues/4247

When installing the APK on Alpine 3.12 armv7 :

```bash
~/MR33542_armv7/packages/testing/armv7 # apk add --allow-untrusted elektra-0.9.9-r0.apk
fetch https://dl-cdn.alpinelinux.org/alpine/edge/main/armv7/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/edge/community/armv7/APKINDEX.tar.gz
fetch https://dl-cdn.alpinelinux.org/alpine/edge/testing/armv7/APKINDEX.tar.gz
ERROR: FDB format error (line 29629)
(1/15) Installing libudev-zero (1.0.1-r0)
ERROR: libudev-zero-1.0.1-r0: trying to overwrite usr/lib/libudev.so.1 owned by eudev-3.2.9-r3.
(2/15) Installing http-parser (2.9.4-r0)
(3/15) Installing libssh2 (1.10.0-r2)
(4/15) Installing libgit2 (1.4.2-r0)
(5/15) Installing libelektra (0.9.9-r0)
(6/15) Installing elektra-dump (0.9.9-r0)
(7/15) Installing elektra-list (0.9.9-r0)
(8/15) Installing elektra-spec (0.9.9-r0)
(9/15) Installing elektra-sync (0.9.9-r0)
(10/15) Installing elektra-resolver_fm_hpu_b (0.9.9-r0)
(11/15) Installing elektra (0.9.9-r0)
(12/15) Purging eudev (3.2.9-r3)
(13/15) Purging udev-init-scripts (33-r1)
(14/15) Purging eudev-libs (3.2.9-r3)
(15/15) Purging kmod-libs (27-r0)
Executing busybox-1.32.0-r0.trigger
Executing elektra-0.9.9-r0.trigger
elektra: Switching default resolver to resolver_fm_hpu_b
elektra: Switching default storage to dump
1 error; 187 MiB in 912 packages
```

Then this happens :

```bash
Error loading shared library libudev.so.1: No such file or directory (needed by /usr/lib/libnm.so.0)
Error relocating /usr/lib/libnm.so.0: udev_device_get_parent: symbol not found
Error relocating /usr/lib/libnm.so.0: udev_device_get_property_value: symbol not found
Error relocating /usr/lib/libnm.so.0: udev_new: symbol not found
Error relocating /usr/lib/libnm.so.0: udev_ref: symbol not found
Error relocating /usr/lib/libnm.so.0: udev_device_unref: symbol not found
Error relocating /usr/lib/libnm.so.0: udev_unref: symbol not found
Error relocating /usr/lib/libnm.so.0: udev_device_new_from_subsystem_sysname: symbol not found
```

Your thought on this ? Would a symlink help ?

Regards

illiliti, 2022-03-22:
Not sure how is that possible since [libudev-zero](https://git.alpinelinux.org/aports/tree/community/libudev-zero/APKBUILD#n14) has lower priority than [eudev](https://git.alpinelinux.org/aports/tree/main/eudev/APKBUILD#n15)

Kochise, 2022-03-22:
Any clue how to check that ?

`libudev-zero` ain't even mentioned inside the APKBUILD file :

https://gitlab.alpinelinux.org/alpine/aports/-/blob/53b9164025c120249fdc876ea7188f81b7c80558/testing/elektra/APKBUILD

illiliti, 2022-03-22:
I'm not familiar enough with APKBUILD to tell what's wrong. I recommend opening an issue to aports.

Kochise, 2022-03-23:
Done here https://gitlab.alpinelinux.org/alpine/aports/-/issues/13627

closed 2022-07-06 by illiliti