illiliti/libudev-zero #56: Some questions

hongkongkiwi, 2022-11-09:
I'm wondering what the purpose of this is over something like mdevd, are they complimentary or would I choose mdevd or libudev-zero ?

Another question, I need to install libgudev for another software, can I compile this against libudev-zero so I don't actually have to install udev?

illiliti, 2022-11-09:
mdevd is a device manager. libudev-zero is an implementation of libudev API that is compatible with any device manager, not only systemd and its forks. Together they can replace udev.

I never had to build or install libgudev, so I know nothing. I only know that is glib wrapper over libudev API, so I think it should compile. See also #12

capezotte, 2022-11-28:
libgudev works with libudev-zero on my machine.

vincele, 2024-06-14:
I think this has been answered and can be closed