illiliti/libudev-zero #59: Using libdev-zero on Termux

dsonkin, 2023-06-29:
Hello

I'm trying to compile hid2hci.c ( https://git.osdn.net/view?p=android-x86/external-bluetooth-bluez.git;a=blob;f=tools/hid2hci.c;h=8f060f2580c863f57ca5d474d6aa89b399208c07;hb=refs/heads/r-x86 ) using clang C compiler on Termux. Is it possible to install libdev-zero on Termux with clang C compiler?

Thank you
Dmitriy

illiliti, 2023-06-30:
Hi! Technically should be possible, but I can't tell much beyond that since I've never used termux as a full development environment and especially for compiling stuff like bluez and its tools. You are strictly on your own here.

dsonkin, 2023-07-03:
Hello

I was able to compile libdev-zero on Termux with clang C compiler by running "make CC=clang".  However, during hid2hci.c compilation I'm getting errors: "undefined symbol: udev_device_get_syspath", "undefined symbol: udev_device_get_devpath", etc. (It looks like compiler is able to find udev.h.) Any idea on potential reason(s) for these errors?

Thank you
Dmitriy

illiliti, 2023-07-04:
Most likely that's because the directory where libudev-zero library is installed is unknown to linker. One way to fix that is to specify it in LD_LIBRARY_PATH, but I'd recommend consulting termux documentation to seek proper fix.

dsonkin, 2023-07-04:
Hello

I've used "make PREFIX=/data/data/com.termux/files/usr install", which uses directory specified in termux documentation. If I run clang compiler with -v option I can see L/data/data/com.termux/files/usr install/lib. (In libudev-zero Makefile I had to change AR=ar to AR=llvm-ar.) What would be the good way to test that libudev-zero was correctly compiled and linked? 

Thank you
Dmitriy

illiliti, 2023-07-05:
Check that libudev.a and/or libudev.so exist in /data/data/com.termux/files/usr/lib. If they are present, then libudev-zero was successfully compiled and installed.

Also before compiling hid2hci.c ensure that `-ludev` option will be passed to compiler. Its absense could be a reason why you get undefined symbol errors.

dsonkin, 2023-07-05:
Thank you for your help, passing -ludev option to compiler, has resolved these errors.

Best,
Dmitriy

closed 2023-07-05 by dsonkin