illiliti/libudev-zero #60: libgudev 238 build failure

pg83, 2023-07-06:
```
meson.build:48:14: ERROR: Dependency lookup for libudev with method 'pkgconfig' failed: Invalid version, need 'libudev' ['>= 251'] found '243'.
```

or, if manually patch libgudev's meson.build:

```
[2/7] Compiling C object gudev/libgudev-1.0.a.p/gudevdevice.c.o
FAILED: gudev/libgudev-1.0.a.p/gudevdevice.c.o 
clang -Igudev/libgudev-1.0.a.p -Igudev -I../src/gudev -I. -I../src -I/ix/store/uLZwZLFmS7MUKLco-lib-glib-ix/include/glib-2.0 -I/ix/store/uLZwZLFmS7MUKLco-lib-glib-ix/lib/glib-2.0/include -I/ix/store/GPIWf4WiE7ZIx2wg-lib-pcre-2/include -I/ix/store/uLZwZLFmS7MUKLco-lib-glib-ix/include -I/ix/store/ZufDx1e6EVfFCuNi-lib-ffi/include -I/ix/store/IZd80D0arFljmJVh-lib-udev-zero/include -fcolor-diagnostics -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -O2 -g -Wno-unknown-warning-option -fPIC -pthread -DHAVE_CONFIG_H -D_POSIX_PTHREAD_SEMANTICS -D_REENTRANT -D_GUDEV_COMPILATION -D_GUDEV_WORK_AROUND_DEV_T_BUG '-DG_LOG_DOMAIN="GUdev"' -MD -MQ gudev/libgudev-1.0.a.p/gudevdevice.c.o -MF gudev/libgudev-1.0.a.p/gudevdevice.c.o.d -o gudev/libgudev-1.0.a.p/gudevdevice.c.o -c ../src/gudev/gudevdevice.c
../src/gudev/gudevdevice.c:146:12: error: call to undeclared function 'udev_device_get_current_tags_list_entry'; ISO C99 and later do not support implicit function declarations [-Wimplicit-function-declaration]
  for (l = udev_device_get_current_tags_list_entry (device->priv->udevice); l != NULL; l = udev_list_entry_get_next (l))
           ^
../src/gudev/gudevdevice.c:146:12: note: did you mean 'udev_device_get_tags_list_entry'?
/ix/store/IZd80D0arFljmJVh-lib-udev-zero/include/libudev.h:72:25: note: 'udev_device_get_tags_list_entry' declared here
struct udev_list_entry *udev_device_get_tags_list_entry(struct udev_device *udev_device);
                        ^
1 error generated.
```

illiliti, 2023-07-07:
Simply adding a stub of udev_device_get_current_tags_list_entry should be enough to unbreak the build. I'll push this fix and also bump version number when I get home.

illiliti, 2023-07-10:
99f1bc955a26eef8c9f4f5fee3247571618351b8
ee32ac5f6494047b9ece26e7a5920650cdf46655

I can cut a new release for you. Interested?

illiliti, 2023-07-13:
Done

closed 2023-07-13 by illiliti