illiliti/libudev-zero #65: Building on macOS fails due to using Linux-only headers

barracuda156, 2024-01-24:
A quick attempt to build this on macOS reveals that it unconditionally includes Linux-only headers.
Is it possible to fix it?

```
/opt/local/var/macports/build/_opt_svacchanda_SonomaPorts_devel_libudev-zero/libudev-zero/work/compwrap/cc/usr/bin/clang -I/opt/local/include -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -Os -isysroot/Library/Developer/CommandLineTools/SDKs/MacOSX14.sdk -arch arm64 -std=c99 -fPIC -pthread -D_XOPEN_SOURCE=700 -Wall -Wextra -Wpedantic -Wmissing-prototypes -Wstrict-prototypes -Wno-unused-parameter -c -o udev_monitor.o udev_monitor.c
In file included from udev.c:20:
./udev.h:23:10: fatal error: 'sys/sysmacros.h' file not found
#include <sys/sysmacros.h>
         ^~~~~~~~~~~~~~~~~
In file included from udev_list.c:21:
./udev.h:23:10: fatal error: 'sys/sysmacros.h' file not found
#include <sys/sysmacros.h>
         ^~~~~~~~~~~~~~~~~
udev_monitor.c:22:10: fatal error: 'linux/netlink.h' file not found
#include <linux/netlink.h>
         ^~~~~~~~~~~~~~~~~
1 error generated.
1 error generated.
1 error generated.
udev_device.c:24:10: fatal error: 'linux/input.h' file not found
#include <linux/input.h>
         ^~~~~~~~~~~~~~~
make: *** [udev_monitor.o] Error 1
make: *** Waiting for unfinished jobs....
make: *** [udev.o] Error 1
make: *** [udev_list.o] Error 1
```

illiliti, 2024-01-24:
No, this library is strictly linux-only.

closed 2024-01-24 by illiliti

barracuda156, 2024-01-24:
Got it, thanks for responding.