illiliti/libudev-zero #68: Why CUPS doesn't work

Steve-at-Ford-Lodge, 2024-02-03:
I compiled and installed the library on my system and found that printing no longer worked. This is because my printer is USB and **libusb** fails, complaining about the lack of version information in libudev.
(Similarly, gparted will not run, with the same issue. Both CUPS and gparted work with _no_ libudev installed.)

illiliti, 2024-02-03:
It doesn't work because libudev-zero doesn't come with symbol versioning by default anymore, which presence is necessary if you want to replace libudev with libudev-zero without rebuilding every package that depends on it. See https://github.com/illiliti/libudev-zero/commit/000ff7bf2fa463ea99afaea8503c2ed9f8a1852c and its follow-up https://github.com/illiliti/libudev-zero/pull/40 for direction how to enable symbol versioning. Perhaps you will also need to update libudev.sym since systemd is still adding new symbols to it.

Steve-at-Ford-Lodge, 2024-02-04:
Thank you. That worked.

closed 2024-02-04 by Steve-at-Ford-Lodge

sertonix, 2024-02-04:
The README still states `cups - needs investigation`. Could that may be changed if the reason is known?